chinezbrun/pytes_serial #7: mqtt home assistant

hurly85, 2023-01-17:
Hi, thanks for your great script, i got it working on my raspberry pi. The only thing i am struggling with is the mqtt connection to home assistant. In the pytes_serial.cfg i set mqtt to active and i added the local ip address of my home assistant server.
In the script i get the error: mqtt publish errorConnection Refused: not authorised
How or where can i add my mqtt broker credentials to your script?

chinezbrun, 2023-01-17:
you need to have an MQTT_broker up and running (i.e Mosquito). Home Assistant is a MQTT client.   I have Eclipse Mosquito installed on my Synology NAS. You can try also Home assistant add on here [Mosquitto](url) -- never used or alternanive for Raspberry PI.

hurly85, 2023-01-19:
Thanks for your answer, i have mosquito integration installed on home assistant and more then 20 devices connected. The issue is that my home assistant mqtt integration requests user and password when a mqtt device tries to connect. I will test it with a another mqtt broker as i am playing arround with my mqtt integration on home assistant and i can´t get anonymous mqtt devices to connect. Do you use your mqtt broker without authentication?

chinezbrun, 2023-01-19:
yes i'm using mqtt broker w/o authentication. all tests were done without authentication.

closed 2023-01-23 by chinezbrun

hurly85, 2023-01-26:
Thanks sorry for my absense, yes i will try it

hurly85, 2023-01-26:
Hi, i tested it and it worked, thanks a lot

chinezbrun, 2023-01-26:
thanks of spot it out this opportunity. please confirm the pytes_serial version used.

hurly85, 2023-01-27:
Great work the script is running all night without error, see pics attached
[image]
[image]

chinezbrun, 2023-01-27:
super!

commit 9445cd52 (2023-01-23): bump to 0.3.0

fix the incompatibility issue with Pytes 'C' models -- closes #9
introduced mqtt authentication (better compatibility to Home Assistant) -- closes #7
introduced statistics -- i.e overall info for the bank -- closes #10
small refactoring