chinezbrun/pytes_serial #26: With the Pylontech US2000C the cell values are not read out.

001Tycoon, 2024-04-29:
Hello,
I have two Pylontech US2000C via pytes_serial.py, the cell values are not read

Device address      : 1
Manufacturer        : Pylon
Device name         : US2000C
Board version       : V10R04
Board               : NF4.E2
Main Soft version   : B69.18.0.0
Soft  version       : V2.1
Boot  version       : V1.0
Comm version        : V2.0
Release Date        : 23-06-02
Specification       : 48V/50AH
Cell Number         : 15
Max Dischg Curr     : -90000mA
Max Charge Curr     : 90000mA
EPONPort rate       : 1200
Console Port rate   : 115200

Device address      : 2
Manufacturer        : Pylon
Device name         : US2000C
Board version       : V10R04
Board               : NF4.E2
Main Soft version   : B69.18.0.0
Soft  version       : V2.1
Boot  version       : V1.0
Comm version        : V2.0
Release Date        : 23-06-02
Specification       : 48V/50AH
Cell Number         : 15
Max Dischg Curr     : -90000mA
Max Charge Curr     : 90000mA
EPONPort rate       : 1200
Console Port rate   : 115200

battery_events.log
|    INFO| START - PytesSerial v0.7.0_20240427
|    INFO| CHECK EVENTS - power_1 cells details:cells data could not be read
|    INFO| START - PytesSerial v0.7.0_20240427

pytes_serial.log
|    INFO| START - PytesSerial v0.7.0_20240427
|    INFO| PARSING BAT - error handling message: invalid literal for int() with base 10: '15%'
|    INFO| PARSING BAT - error handling message: invalid literal for int() with base 10: '16%'
|    INFO| PARSING BAT - error handling message: invalid literal for int() with base 10: '15%'

Outputs via the console
[Pylontech.txt](https://github.com/chinezbrun/pytes_serial/files/15146437/Pylontech.txt)

Thanks for the tool

chinezbrun, 2024-04-29:
Do you have the sensors in HA for cells monitoring? If yes there is any data there?

001Tycoon, 2024-04-29:
Hello

created for both batteries. but they are all unknown.

pylontech_cells curr_st_

pylontech_cells soc_

pylontech_cells status_

pylontech_cells temp_st_

pylontech_cells temperature_

pylontech_cells temperature_delta_

pylontech_cells temperature_max_

pylontech_cells temperature_min_

pylontech_cells volt_st_

pylontech_cells voltage_

pylontech_cells voltage_delta_

pylontech_cells voltage_max_

pylontech_cells voltage_min_

```
pylon>bat 1
@
Battery  Volt     Curr     Tempr    Base State   Volt. State  Curr. State  Temp. State  SOC          Coulomb      BAL
0        3286     7303     14000    Charge       Normal       Normal       Normal       19%          9705 mAH      N
1        3285     7303     14000    Charge       Normal       Normal       Normal       19%          9705 mAH      N
2        3285     7303     14000    Charge       Normal       Normal       Normal       19%          9705 mAH      N
3        3289     7303     14000    Charge       Normal       Normal       Normal       19%          9705 mAH      N
4        3288     7303     14000    Charge       Normal       Normal       Normal       19%          9705 mAH      N
5        3287     7303     13800    Charge       Normal       Normal       Normal       19%          9705 mAH      N
6        3284     7303     13800    Charge       Normal       Normal       Normal       19%          9705 mAH      N
7        3281     7303     13800    Charge       Normal       Normal       Normal       19%          9705 mAH      N
8        3286     7303     13800    Charge       Normal       Normal       Normal       19%          9705 mAH      N
9        3283     7303     13800    Charge       Normal       Normal       Normal       18%          9215 mAH      N
10       3286     7303     13700    Charge       Normal       Normal       Normal       19%          9705 mAH      N
11       3286     7303     13700    Charge       Normal       Normal       Normal       19%          9705 mAH      N
12       3286     7303     13700    Charge       Normal       Normal       Normal       19%          9705 mAH      N
13       3286     7303     13700    Charge       Normal       Normal       Normal       19%          9705 mAH      N
14       3286     7303     13700    Charge       Normal       Normal       Normal       19%          9705 mAH      N
Command completed successfully
$$
```

001Tycoon, 2024-04-29:
This is what it looks like in Home Assistant
[image]

chinezbrun, 2024-04-29:
check the last comit of pytes_serial.py if fixed the issue

001Tycoon, 2024-04-29:
thank you, it works

commit fa506613 (2024-04-30): bump to 0.7.1_20240430

close #28 , close #26
closed 2024-04-29 by 001Tycoon