chinezbrun/pytes_serial #35: some terminal output improvents

stenis25856, 2024-07-30:
Not an issue...
For now, i use pytes_serial only to get sometimes some data from the battery, 
i have four of them, great work :)

I add some features to get a more detailed terminal output, here is one example:

------------------------------------------------------
relay local time: 2024-07-30 09:17:26
serial uptime   : 0.007
...serial parsing: ok
...json creation:  ok

power          1         2         3         4         
voltage        49.74     49.78     49.76     49.77     
current        2.03      2.14      1.87      1.82      
temperature    28.5      29.0      28.0      27.0      
soc            57        53        57        58        
basic_st       Charge    Charge    Charge    Charge    
volt_st        Normal    Normal    Normal    Normal    
current_st     Normal    Normal    Normal    Normal    
temp_st        Normal    Normal    Normal    Normal    
soh_st         Normal    Normal    Normal    Normal    
coul_st        Normal    Normal    Normal    Normal    
heater_st      OFF       OFF       OFF       OFF       
bat_events     0         0         0         0         
power_events   0         0         0         0         
sys_events     0         0         0         0         
V_delta        0.001     0.005     0.003     0.003     
T_delta        0.4       1.0       1.0       0.0       

3.316 3.315 3.315 3.316 3.316 3.316 3.316 3.315 3.315 3.316 3.316 3.316 3.316 3.316 3.316 | Sum: 49.736
3.318 3.318 3.319 3.318 3.317 3.316 3.317 3.317 3.318 3.320 3.319 3.321 3.320 3.320 3.318 | Sum: 49.776
3.316 3.317 3.317 3.316 3.317 3.318 3.317 3.318 3.317 3.317 3.318 3.319 3.318 3.319 3.318 | Sum: 49.762
3.317 3.319 3.319 3.318 3.317 3.318 3.317 3.316 3.317 3.317 3.317 3.318 3.318 3.318 3.317 | Sum: 49.763

27.97 27.97 27.98 27.97 27.95 27.98 27.98 27.98 27.98 27.98 27.98 27.98 27.97 27.95 27.97 | Sum: 419.589
24.43 24.43 24.43 24.43 24.43 24.43 24.43 24.43 24.43 24.43 24.43 24.43 24.43 24.43 24.43 | Sum: 366.465
26.37 26.37 26.37 26.37 26.37 26.37 26.37 26.37 26.37 26.37 26.37 26.37 26.37 26.37 26.37 | Sum: 395.550
26.78 26.78 26.78 26.78 26.78 26.78 26.78 26.78 26.78 26.78 26.78 26.78 26.78 26.78 26.78 | Sum: 401.715
...serial stat   : loops: 60 errors: 7 efficiency: 88.33
...serial stat   : bat events_no: 0 pwr events_no: 0 sys events_no: 0
...serial stat   : parsing round-trip: 1.57
------------------------------------------------------

And here is the code, i added it at line 457 (code is fixed for 4 batteries and 15 cells):

        # load Json
        data = json_data
        
        # Output Devices
        device_headers = ["power", "voltage", "current", "temperature", "soc", "basic_st", "volt_st", "current_st", "temp_st", "soh_st", "coul_st", "heater_st", "bat_events", "power_events", "sys_events", "V_delta", "T_delta"]

        device_matrix = []
        
        print()

        for device in data['devices']:
            device_row = [device[header] for header in device_headers[:15]]
            cells_data = next((cd for cd in data['cells_data'] if cd['power'] == device['power']), {})
            device_row.append(cells_data.get('voltage_delta', None))
            device_row.append(cells_data.get('temperature_delta', None))
            device_matrix.append(device_row)

        device_matrix_transposed = list(map(list, zip(*device_matrix)))

        for i, header in enumerate(device_headers):
            print(f"{header:<15}", end="")
            for device in device_matrix_transposed[i]:
                print(f"{device:<10}", end="")
            print()
        
        print()

        #Output cell voltages
        voltage_matrix = np.full((4, 15), None)

        # voltages in matrix
        for device in data['cells_data']:
            power = device['power'] - 1  
            for cell in device['cells']:
                cell_index = cell['cell'] - 1  
                voltage_matrix[power, cell_index] = cell['voltage']

        # Output in wide format
        for row in voltage_matrix:
            row_sum = sum(v for v in row if v is not None)  # sum of voltages
            print(" ".join(f"{v:.3f}" if v is not None else "None" for v in row), f"| Sum: {row_sum:.3f}")
      
        print()
        
        #Output cell Coulomb
        coulomb_matrix = np.full((4, 15), None)

        # coulomb in matrix
        for device in data['cells_data']:
            power = device['power'] - 1  
            for cell in device['cells']:
                cell_index = cell['cell'] - 1  
                coulomb_matrix[power, cell_index] = cell['coulomb']

        # Output in wide format
        for row in coulomb_matrix:
            row_sum = sum(v for v in row if v is not None)  # sum of coulomb
            print(" ".join(f"{v:.2f}" if v is not None else "None" for v in row), f"| Sum: {row_sum:.3f}")

regards
 stefan

chinezbrun, 2024-07-30:
Thanks for sharing and your feedback! 
The console output was designed more for status checks / debug rather than monitoring purpose. Therefore, it was a conscious decision to have the console output per battery in the parsing sequence and clean as much as possible. I addition to that with the cells monitoring active when a critical event is happening, the individual cells data are displayed too on the console.

chinezbrun, 2024-08-14:
no further info action needed

closed 2024-08-14 by chinezbrun