chinezbrun/pytes_serial #36: Feature request - state of health

andi-at-1, 2024-09-19:
Hello,

You have made a great tool!
But may i ask, could you implement the state of health command?

It is read in hyperterminal by "soh x" (battery count), and is calculated by current capacity / original capacity in percent. 
I am at 100% so i dunno what changes when the health drops, but i am sure someone with longer battery runtime will see different values.

example output at 100% health:
`soh 1
@          
Power   1
Battery    Voltage    SOHCount   SOHStatus 
0          3361       00000000
1          3361       00000000
2          3361       00000000
3          3362       00000000
4          3364       00000000
5          3364       00000000
6          3363       00000000
7          3364       00000000
8          3370       00000000
9          3365       00000000
10         3366       00000000
11         3369       00000000
12         3366       00000000
13         3366       00000000
14         3365       00000000
15         3365       00000000
Command completed successfully
$$
PYTES>soh 2
@          
Power   2
Battery    Voltage    SOHCount   SOHStatus 
0          3352       00000000
1          3350       00000000
2          3352       00000000
3          3352       00000000
4          3352       00000000
5          3351       00000000
6          3352       00000000
7          3351       00000000
8          3355       00000000
9          3351       00000000
10         3353       00000000
11         3354       00000000
12         3353       00000000
13         3352       00000000
14         3352       00000000
15         3351       00000000
Command completed successfully
$$
PYTES>`

thank you!

chinezbrun, 2024-09-20:
Thank for your feedback. May i ask what will be the value added?

andi-at-1, 2024-09-21:
Honestly i dunno how good this calculation is, but if its somewhat correct you will be aware if a cell or a whole battery will see a drop in capacity, and therefore get an idea if you should change some unhealthy settings like reduce discarging, improve cooling, etc. way before you damage your battery seriously.

chinezbrun, 2024-09-21:
I understood. The main challenge is not to implement another command but to maintain the efficiency of the program. For example, when cell monitoring is "active", both "pwr" and "bat" commands are sent. The average round-trip time is ~ 0.5 s/battery. In this case, for a bank of 8 batteries, the total round trip becomes ~4 seconds. Any additional command will increase the round trip beyond 5 seconds and this is unacceptable as it will be a source of errors.

andi-at-1, 2024-09-23:
oh ok that makes sense. The value won´t change fast, so would it be possible to trigger the reading lets say once a day/week? Or grab it just at the start of the script, so i can just kill it from time to time?

chinezbrun, 2024-10-06:
Despite the fact is not a priority, I will leave the issue open for a while to collect other opinions/votes :). 
If I'll decided to implement #37 then, based on other users votes i will think about 'SoH' too :)

chinezbrun, 2024-11-10:
I will now close this issue as it is unlikely to be implemented in the near future. Thank you for the suggestion and even if it was not implemented I hope you like the new version of the program.

closed 2024-11-10 by chinezbrun